Open-X-Humanoid/TienKung-Lab #1: is the actor obs  in sim2sim and real robot need linear vel

dbdxnuliba, 2025-07-08:
thanks for your execellent works, but we can see the code in sim2sim.py
[image]

is the actor obs  in sim2sim and real robot need linear vel

BryceWang0527, 2025-07-08:
Yes, in this framework, the actor's observation does include linear velocity.

- In simulation, the linear velocity is directly obtained from the ground-truth provided by the physics engine.

- On the real robot, the linear velocity is replaced by a state-estimated value.

Thus, although the data source differs (ground truth vs. estimation), the observation format remains consistent.

dbdxnuliba, 2025-07-09:
thanks , could you please provide the   state-estimated module

sixtheta-j, 2025-07-10:
> > 实机必须要状态估计吗？不加不可以吗
> 
> 不需要 — 此存储库中确认的原始 LeggedLab 存储库也不包含线性速度，它仍然可以在机器人上运行。
> 
> <img alt="图像" width="634" height="127" src="https://private-user-images.githubusercontent.com/171106827/464480416-6d2adda6-6084-418c-8ae7-010f5e989650.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIxMjYzODIsIm5iZiI6MTc1MjEyNjA4MiwicGF0aCI6Ii8xNzExMDY4MjcvNDY0NDgwNDE2LTZkMmFkZGE2LTYwODQtNDE4Yy04YWU3LTAxMGY1ZTk4OTY1MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcxMFQwNTQxMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jOTM1NmY1ZjJiZTBiMGNkNzc5ZmQ3NmYzYmY4ZjhhNjBiMjY2OWNiOWViY2M5ZDlhMjM0ZmJhODgyOWEyN2JlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.geKwS6zTJjoA3OvO7ZS2oUl8tLS4LkwTnf8k0xMf5Do">

谢谢

closed 2025-07-11 by soulferryman