Open-X-Humanoid/TienKung-Lab #15: .xml question

QingZhuanya, 2025-08-01:
May I ask how you obtained the .xml files in your MJCF models? When I export a .xml file from my robot’s URDF, I always get errors.

```
Traceback (most recent call last):
  File "/home/robot/robot/TienKung-Lab/legged_lab/scripts/sim2sim.py", line 347, in <module>
    runner = MujocoRunner(
  File "/home/robot/robot/TienKung-Lab/legged_lab/scripts/sim2sim.py", line 69, in __init__
    self.model = mujoco.MjModel.from_xml_path(model_path)
ValueError: Error: empty name in material
Element name '', id 0, line 21
```

BryceWang0527, 2025-08-04:
It looks like the error comes from an empty name field in one of the <material> tags in your converted MJCF file, you can try the following repo.

[URDF2MJCF](https://github.com/FFTAI/Wiki-GRx-MJCF)

QingZhuanya, 2025-08-04:
> It looks like the error comes from an empty name field in one of the tags in your converted MJCF file, you can try the following repo.
> 
> [URDF2MJCF](https://github.com/FFTAI/Wiki-GRx-MJCF)

Thank you for your reply。I'm using a Tienkung Pro robot (with 30 joints in the urdf file). I'm experiencing issues with the robot moving normally in Isaac Sim, but failing in Mujoco. I noticed that the .pt file only controls 20 joints. How can I modify the train.py file or sim2sim.py file so that the Pro version's 20 joints match the Lite version's 20 joints, with the remaining joints set to 0?

BryceWang0527, 2025-08-06:
Great to hear that the training works well in Isaac Sim. Since model only controls 20 joints (Lite version), you can modify the part of the sim2sim.py where joint actions are sent to MuJoCo.

Specifically, in the line:
`self.data.ctrl = self.position_control()`

You can keep the network's input and output unchanged, and manually set the remaining 10 joints (those not controlled by the policy) to zero. This way, the 20 joints will be driven by the trained policy, and the additional 10 joints (Pro version) will be fixed at zero position command.

QingZhuanya, 2025-09-03:
> s

Thank you, this issue has been resolved. May I ask if the pro version of the sim2real code will be open-sourced?

QingZhuanya, 2025-09-03:
> Thank you for your question! The AMP data itself will not affect the sim2sim transfer in MuJoCo — it only matters for training. If you could share videos from both Isaac Sim and MuJoCo, I would be able to get a clearer picture of the issue. My guess is that the problem may come from a mismatch in the joint ordering between Isaac Sim and MuJoCo.

Thank you, this issue has been resolved. May I ask if the pro version of the sim2real code will be open-sourced?

closed 2025-09-08 by soulferryman